PX4/PX4-Avoidance #680: Process died error local planner:Implementing local_planner on hardware

Samyak-ja-in, 2022-07-20:
@Jaeyoung-Lim I am trying to implement local planner on hardware while launching avoidance.launch file I got following errors
also on running command 
```
rostopic hz /local_pointcloud
```
it shows no new messages
```
[ INFO] [1658155662.107217276]: Plugin waypoint initialized
[ INFO] [1658155662.107429369]: Plugin wheel_odometry blacklisted
[ INFO] [1658155662.108138647]: Plugin wind_estimation loaded
[ INFO] [1658155662.111820443]: Plugin wind_estimation initialized
[ INFO] [1658155662.112520073]: Autostarting mavlink via USB on PX4
[ INFO] [1658155662.113023443]: Built-in SIMD instructions: None
[ INFO] [1658155662.113175536]: Built-in MAVLink package version: 2021.3.3
[ INFO] [1658155662.113333869]: Known MAVLink dialects: common ardupilotmega ASLUAV all autoquad icarous matrixpilot paparazzi standard uAvionix ualberta
[ INFO] [1658155662.113529147]: MAVROS started. MY ID 1.240, TARGET ID 1.1
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/dynamic_reconfigure/dynparam", line 240, in <module>
    elif cmd == 'load':                do_load()
  File "/opt/ros/melodic/lib/dynamic_reconfigure/dynparam", line 159, in do_load
    f = file(path, 'r')
NameError: name 'file' is not defined
[dynparam-9] process has died [pid 16503, exit code 1, cmd /opt/ros/melodic/lib/dynamic_reconfigure/dynparam load local_planner_node /home/pi/catkin_ws/src/avoidance/local_planner/cfg/params_vehicle_1.yaml __name:=dynparam __log:=/home/pi/.ros/log/8ff890d0-06a8-11ed-839a-e45f01b3f146/dynparam-9.log].
log file: /home/pi/.ros/log/8ff890d0-06a8-11ed-839a-e45f01b3f146/dynparam-9*.log
[ INFO] [1658155663.114383850]: Device with serial number 033422072382 was found.

[ INFO] [1658155663.114520202]: Device with physical ID 2-2.2-4 was found.
[ INFO] [1658155663.114590035]: Device with name Intel RealSense D435I was found.
[ INFO] [1658155663.116653609]: Device with port number 2-2.2 was found.
[ERROR] [1658155663.125840813]: The requested device with serial number 032523050787 is NOT found. Will Try again.
Waiting for service /camera_main/stereo_module/set_parameters...
[ INFO] [1658155667.544646181]: Initializing nodelet with 4 worker threads.
[ERROR] [1658155667.921101773]: PR: Unknown parameter to get: MPC_ACC_HOR
[ERROR] [1658155667.983632291]: PR: Unknown parameter to get: MPC_ACC_DOWN_MAX
[ERROR] [1658155667.989028180]: PR: Unknown parameter to get: MPC_ACC_UP_MAX
[ERROR] [1658155667.993614736]: PR: Unknown parameter to get: MPC_XY_CRUISE
[ERROR] [1658155667.996858236]: PR: Unknown parameter to get: MPC_Z_VEL_MAX_DN
[ERROR] [1658155668.001532180]: PR: Unknown parameter to get: MPC_Z_VEL_MAX_UP
[ERROR] [1658155668.004661940]: PR: Unknown parameter to get: CP_DIST
[ERROR] [1658155668.007742291]: PR: Unknown parameter to get: MPC_LAND_SPEED
[ERROR] [1658155668.011153495]: PR: Unknown parameter to get: MPC_JERK_MAX
[ERROR] [1658155668.014762440]: PR: Unknown parameter to get: NAV_ACC_RAD
[ERROR] [1658155668.017938458]: PR: Unknown parameter to get: MPC_YAWRAUTO_MAX

```
Thanks!

Samyak-ja-in, 2022-07-20:
The above error is solved just by defining fcu port during launching and launching as
```
roslaunch local_planner avoidance.launch fcu_url:=/dev/ttyACM0:57600
```
but now getting error that Intel RealSense D435I device found and again after some seconds device not found
```
[ INFO] [1658155849.882075048]: Device with physical ID 2-2.3-4 was found.
[ INFO] [1658155849.882548085]: Device with name Intel RealSense D435I was found.
[ INFO] [1658155849.884746048]: Device with port number 2-2.3 was found.
[ERROR] [1658155849.888126789]: The requested device with serial number 032523050787 is NOT found. Will Try again.
[ WARN] [1658155855.241019841]: CMD: Unexpected command 410, result 2
[ INFO] [1658155855.987717285]:  
[ INFO] [1658155856.678939211]: Device with serial number 033422072382 was found.

[ INFO] [1658155856.679062266]: Device with physical ID 2-2.3-4 was found.
[ INFO] [1658155856.679124155]: Device with name Intel RealSense D435I was found.
[ INFO] [1658155856.680348415]: Device with port number 2-2.3 was found.
```
Thanks!

Jaeyoung-Lim, 2022-07-20:
@Samyak-ja-in I see a few things
a) `fcu_url:=/dev/ttyACM0:57600` is way too low to do anything what this repository is doing. The baudrate is simply too low.

b) You would need to debug why your device is keep getting rediscovered. If it is a power issue that it is getting restarted periodically or there is something with the device management on the systems side. If you think the issue is related to something in this repository, please report back and I will try to help.

Samyak-ja-in, 2022-07-22:
@Jaeyoung-Lim Have resolved the too low baud rate by simply increasing it
But actually the realsense device only restarts periodically when I launch avoidance.launch launch file, it behaves normally when I view its feed via realsense-viewer
Here is my launch file
```
<launch>
    <arg name="ns" default="/"/>
    <arg name="fcu_url" default="udp://:14540@localhost:14557"/>
    <arg name="gcs_url" default="" />   <!-- GCS link is provided by SITL -->
    <arg name="tgt_system" default="1" />
    <arg name="tgt_component" default="1" />

    <!-- Launch MavROS -->
    <group ns="$(arg ns)">
        <include file="$(find mavros)/launch/node.launch">
            <arg name="pluginlists_yaml" value="$(find mavros)/launch/px4_pluginlists.yaml" />
            <!-- Need to change the config file to get the tf topic and get local position in terms of local origin -->
            <arg name="config_yaml" value="$(find local_planner)/resource/px4_config.yaml" />
            <arg name="fcu_url" value="$(arg fcu_url)" />
            <arg name="gcs_url" value="$(arg gcs_url)" />
            <arg name="tgt_system" value="$(arg tgt_system)" />
            <arg name="tgt_component" value="$(arg tgt_component)" />
        </include>
    </group>

    <!-- Launch cameras -->
<node pkg="tf" type="static_transform_publisher" name="tf_camera_main" required="true"
 args="0.3 0.32 -0.11 0 0 0 fcu camera_main_link 10"/>
<include file="$(find local_planner)/launch/rs_depthcloud.launch">
<arg name="required"              value="true"/>
<arg name="namespace"             value="camera_main" />
<arg name="tf_prefix"             value="camera_main" />
<arg name="serial_no"             value="032523050787"/>
<arg name="depth_fps"             value="30"/>
</include>

      <!-- launch node to throttle depth images for logging -->
      <node name="drop_camera_main_depth" pkg="topic_tools" type="drop" output="screen"
        args="/camera_main/depth/image_rect_raw 29 30">
      </node>
      <node name="drop_camera_main_ir" pkg="topic_tools" type="drop" output="screen"
        args="/camera_main/infra1/image_rect_raw 29 30">
      </node>
    <node name="dynparam" pkg="dynamic_reconfigure" type="dynparam" args="load local_planner_node /home/pi/catkin_ws/src/avoidance/local_planner/cfg/params_vehicle_1.yaml" />
    <!-- Launch avoidance -->
    <env name="ROSCONSOLE_CONFIG_FILE" value="$(find local_planner)/resource/custom_rosconsole.conf"/>
    <arg name="pointcloud_topics" default="[/camera_main/depth/points]"/>

    <node name="local_planner_node" pkg="local_planner" type="local_planner_node" output="screen" required="true" >
      <param name="goal_x_param" value="0" />
      <param name="goal_y_param" value="0"/>
      <param name="goal_z_param" value="4" />
      <rosparam param="pointcloud_topics" subst_value="True">$(arg pointcloud_topics)</rosparam>
    </node>
    
    <!-- set or toggle rqt parameters -->
    <node name="rqt_param_toggle" pkg="local_planner" type="rqt_param_toggle.sh" />

</launch>

```

Samyak-ja-in, 2022-07-22:
Also getting this process died error @Jaeyoung-Lim 
```
[ INFO] [1658400924.359154265]: Plugin vfr_hud initialized
[ INFO] [1658400924.359912691]: Plugin vibration blacklisted
[ INFO] [1658400924.361160710]: Plugin vision_pose_estimate loaded
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/dynamic_reconfigure/dynparam", line 240, in <module>
    elif cmd == 'load':                do_load()
  File "/opt/ros/melodic/lib/dynamic_reconfigure/dynparam", line 159, in do_load
    f = file(path, 'r')
NameError: name 'file' is not defined
[ INFO] [1658400924.420038080]: Plugin vision_pose_estimate initialized
[ INFO] [1658400924.420819950]: Plugin vision_speed_estimate loaded
[ INFO] [1658400924.450337043]: Plugin vision_speed_estimate initialized
[ INFO] [1658400924.451240599]: Plugin waypoint loaded
[ INFO] [1658400924.492150191]: Plugin waypoint initialized
[ INFO] [1658400924.492319099]: Plugin wheel_odometry blacklisted
[ INFO] [1658400924.492884636]: Plugin wind_estimation loaded
[ INFO] [1658400924.496347895]: Plugin wind_estimation initialized
[ INFO] [1658400924.497170802]: Autostarting mavlink via USB on PX4
[ INFO] [1658400924.497700136]: Built-in SIMD instructions: None
[ INFO] [1658400924.497943895]: Built-in MAVLink package version: 2021.3.3
[ INFO] [1658400924.498134728]: Known MAVLink dialects: common ardupilotmega ASLUAV all autoquad icarous matrixpilot paparazzi standard uAvionix ualberta
[ INFO] [1658400924.498290524]: MAVROS started. MY ID 1.240, TARGET ID 1.1
[ INFO] [1658400924.500390043]: IMU: Scaled IMU message used.
[ INFO] [1658400924.509327913]: IMU: Attitude quaternion IMU detected!
[ INFO] [1658400924.510752173]: IMU: High resolution IMU detected!
[dynparam-9] process has died [pid 8544, exit code 1, cmd /opt/ros/melodic/lib/dynamic_reconfigure/dynparam load local_planner_node /home/pi/catkin_ws/src/avoidance/local_planner/cfg/params_vehicle_1.yaml __name:=dynparam __log:=/home/pi/.ros/log/9bdd29e6-08e3-11ed-bc36-e45f01b3f146/dynparam-9.log].
log file: /home/pi/.ros/log/9bdd29e6-08e3-11ed-bc36-e45f01b3f146/dynparam-9*.log

```

Samyak-ja-in, 2022-07-22:
Also getting this process died error it was not solved as mentioned above @Jaeyoung-Lim 
```
[ INFO] [1658400924.359154265]: Plugin vfr_hud initialized
[ INFO] [1658400924.359912691]: Plugin vibration blacklisted
[ INFO] [1658400924.361160710]: Plugin vision_pose_estimate loaded
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/dynamic_reconfigure/dynparam", line 240, in <module>
    elif cmd == 'load':                do_load()
  File "/opt/ros/melodic/lib/dynamic_reconfigure/dynparam", line 159, in do_load
    f = file(path, 'r')
NameError: name 'file' is not defined
[ INFO] [1658400924.420038080]: Plugin vision_pose_estimate initialized
[ INFO] [1658400924.420819950]: Plugin vision_speed_estimate loaded
[ INFO] [1658400924.450337043]: Plugin vision_speed_estimate initialized
[ INFO] [1658400924.451240599]: Plugin waypoint loaded
[ INFO] [1658400924.492150191]: Plugin waypoint initialized
[ INFO] [1658400924.492319099]: Plugin wheel_odometry blacklisted
[ INFO] [1658400924.492884636]: Plugin wind_estimation loaded
[ INFO] [1658400924.496347895]: Plugin wind_estimation initialized
[ INFO] [1658400924.497170802]: Autostarting mavlink via USB on PX4
[ INFO] [1658400924.497700136]: Built-in SIMD instructions: None
[ INFO] [1658400924.497943895]: Built-in MAVLink package version: 2021.3.3
[ INFO] [1658400924.498134728]: Known MAVLink dialects: common ardupilotmega ASLUAV all autoquad icarous matrixpilot paparazzi standard uAvionix ualberta
[ INFO] [1658400924.498290524]: MAVROS started. MY ID 1.240, TARGET ID 1.1
[ INFO] [1658400924.500390043]: IMU: Scaled IMU message used.
[ INFO] [1658400924.509327913]: IMU: Attitude quaternion IMU detected!
[ INFO] [1658400924.510752173]: IMU: High resolution IMU detected!
[dynparam-9] process has died [pid 8544, exit code 1, cmd /opt/ros/melodic/lib/dynamic_reconfigure/dynparam load local_planner_node /home/pi/catkin_ws/src/avoidance/local_planner/cfg/params_vehicle_1.yaml __name:=dynparam __log:=/home/pi/.ros/log/9bdd29e6-08e3-11ed-bc36-e45f01b3f146/dynparam-9.log].
log file: /home/pi/.ros/log/9bdd29e6-08e3-11ed-bc36-e45f01b3f146/dynparam-9*.log

```

Jaeyoung-Lim, 2022-07-22:
@Samyak-ja-in Is there any chance you are running this package on a arm device? I dont think this has been tested yet

When you speak about the realsense viewer, was this running on the pi?

Samyak-ja-in, 2022-07-22:
Yes I am running package on raspberry pi
Yes realsense-viewer is running on rpi

Jaeyoung-Lim, 2022-07-22:
@Samyak-ja-in I think you would need to figure out why the realsense keep getting disconnected. This package has never been tested / or intended to be used on a raspberry pi

Samyak-ja-in, 2022-07-22:
Local planner shows this debugging info and I am unable to set PX4 parameter as it is not available in qgc
```
[ WARN] [1658398687.862099302]:  Planner abort: missing required data from FCU 
 
[ WARN] [1658398687.862259802]: ----------------------------- Debugging Info -----------------------------
[ WARN] [1658398687.862322839]: Local planner has not received a position from FCU, check the following: 
[ WARN] [1658398687.862394580]: 1. Check cables connecting PX4 autopilot with onboard computer
[ WARN] [1658398687.862477617]: 2. Set PX4 parameter MAV_1_MODE to onbard or external vision
[ WARN] [1658398687.862545561]: 3. Set correct fcu_url in local_planner launch file:
[ WARN] [1658398687.862614154]:    Example direct connection to serial port: /dev/ttyUSB0:921600
[ WARN] [1658398687.862690876]:    Example connection over mavlink router: udp://:14540@localhost:14557
[ WARN] [1658398687.862768172]: --------------------------------------------------------------------------
```

Samyak-ja-in, 2022-07-22:
@Jaeyoung-Lim okay how to give the drone setpoints can I simply run the python script in offboard mode

Samyak-ja-in, 2022-07-24:
Hi @Jaeyoung-Lim I am getting two errors like
```
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/dynamic_reconfigure/dynparam", line 240, in <module>
    elif cmd == 'load':                do_load()
  File "/opt/ros/melodic/lib/dynamic_reconfigure/dynparam", line 159, in do_load
    f = file(path, 'r')
NameError: name 'file' is not defined
.
.
.
.
[ERROR] [1658573798.001268089]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1658573798.006441089]: TF Buffer: could not retrieve requested transform from buffer, unregistered
```
have followed this
https://github.com/PX4/PX4-Avoidance/issues/523#issuecomment-552780923
but still the problem persists could you tell me what to do

closed 2022-07-27 by Samyak-ja-in

bhaskar-glitch, 2023-01-17:
@Samyak-ja-in How did you resolved this issue?? Im also getting the same error saying:
**[ERROR] [1673941528.536289239]: TF Buffer: could not retrieve requested transform from buffer, unregistered**