PX4/PX4-Avoidance #681: TF Buffer: could not retrieve requested transform from buffer, unregistered

MichaelKerley, 2022-08-09:
**Describe the bug**
I’m currently attempting to use my intel real sense D435 camera and PX4 Cube pilot with local planner. The error I am experiencing is 
[ERROR] [1660062552.700789171]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1660062552.705819345]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1660062552.711514289]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1660062552.716710617]: TF Buffer: could not retrieve requested transform from buffer, unregistered

Repeated again and again.

**To Reproduce**
Steps to reproduce the behavior:
To run PX4 avoidance I have connected a NVidia jetson nano to the cube pilot using an USB to FTDI Cable as seen in the picture below.
 

I have connected to the Jetson over SSH and am attempting to launch local planner using the instructions provided on [PX4/PX4-Avoidance: PX4 avoidance ROS node for obstacle detection and avoidance. (github.com)](https://github.com/PX4/PX4-Avoidance).
The command I am currently using is 
roslaunch local_planner avoidance.launch fcu_url:=/dev/ttyUSB4:57600
It is USB4 because I have a sixfab hat connected to the jetson already using the other ports.
To run PX4 avoidance I have connected a NVidia jetson nano to the cube pilot using an USB to FTDI Cable
 

I have connected to the Jetson over SSH and am attempting to launch local planner using the instructions provided on [PX4/PX4-Avoidance: PX4 avoidance ROS node for obstacle detection and avoidance. (github.com)](https://github.com/PX4/PX4-Avoidance).
The command I am currently using is 
roslaunch local_planner avoidance.launch fcu_url:=/dev/ttyUSB4:57600
It is USB4 because I have a sixfab hat connected to the jetson already using the other ports.

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/en/getting_started/flight_reporting.html)) or from the Gazebo simulation (logs are located in `~/.ros/logs/`)
- Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/).
- Share a link to a rosbag showing the problem

started roslaunch server http://michael-desktop:42637/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /camera_main/realsense2_camera/accel_fps: 0
 * /camera_main/realsense2_camera/accel_frame_id: camera_main_accel...
 * /camera_main/realsense2_camera/accel_optical_frame_id: camera_main_accel...
 * /camera_main/realsense2_camera/align_depth: False
 * /camera_main/realsense2_camera/aligned_depth_to_color_frame_id: camera_main_align...
 * /camera_main/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_main_align...
 * /camera_main/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_main_align...
 * /camera_main/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_main_align...
 * /camera_main/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_main_align...
 * /camera_main/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_main_align...
 * /camera_main/realsense2_camera/allow_no_texture_points: False
 * /camera_main/realsense2_camera/base_frame_id: camera_main_link
 * /camera_main/realsense2_camera/calib_odom_file: 
 * /camera_main/realsense2_camera/clip_distance: -1.0
 * /camera_main/realsense2_camera/color_fps: 15
 * /camera_main/realsense2_camera/color_frame_id: camera_main_color...
 * /camera_main/realsense2_camera/color_height: 480
 * /camera_main/realsense2_camera/color_optical_frame_id: camera_main_color...
 * /camera_main/realsense2_camera/color_width: 640
 * /camera_main/realsense2_camera/confidence_fps: 30
 * /camera_main/realsense2_camera/confidence_height: 480
 * /camera_main/realsense2_camera/confidence_width: 640
 * /camera_main/realsense2_camera/depth_fps: 30
 * /camera_main/realsense2_camera/depth_frame_id: camera_main_depth...
 * /camera_main/realsense2_camera/depth_height: 480
 * /camera_main/realsense2_camera/depth_optical_frame_id: camera_main_depth...
 * /camera_main/realsense2_camera/depth_width: 640
 * /camera_main/realsense2_camera/device_type: 
 * /camera_main/realsense2_camera/enable_accel: False
[... 459 lines cut ...]
[ERROR] [1660062553.246836537]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1660062553.252020468]: TF Buffer: could not retrieve requested transform from buffer, unregistered
^C[ERROR] [1660062553.257139761]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1660062553.262282858]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1660062553.267408609]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1660062553.272525559]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1660062553.277618809]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1660062553.282738831]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[local_planner_node-10] killing on exit
[ERROR] [1660062553.288120688]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1660062553.292990856]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[drop_camera_main_depth-7] killing on exit
[mavros-2] killing on exit
[camera_main/realsense2_camera-5] killing on exit
[drop_camera_main_ir-8] killing on exit
[camera_main/points_xyz_sw_registered-6] killing on exit
[camera_main/realsense2_camera_manager-4] killing on exit
[tf_camera_main-3] killing on exit
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Hammadiqbal12, 2022-10-17:
Everything is fine, you need to provide position reference to pixhawk. Please run it when PX4 gets GPS fix.

MichaelKerley, 2022-11-01:
> Everything is fine, you need to provide position reference to pixhawk. Please run it when PX4 gets GPS fix

Have you had similar issues in the past?

Badrivishal, 2022-12-23:
Check whether all the values in the local_posistion/pose topic are being published ie. x y and z. If either one of them is static 0, then you do not have a GPS lock. Wait for GPS lock and try it again.