PX4/PX4-Avoidance #682: ROS2 Foxy?

hwadiadinkra, 2022-09-07:
Hi, 

This is an amazing feature, however, the default support being ROS Melodic is really outdated. We are trying to port this into AirSim, and having this native in ROS2 Foxy would greatly decrease the complexity of our network/node setup. 

When will this feature, the local planner specifically, be ported into ROS2 Foxy? If not, then what would be the suggested route to incorporating this into a ROS2/PX4 build, which is what PX4 is pushing for?

Thank you.

jhiggason, 2023-01-04:
Did you ever get ROS2 and PX4-Avoidance working together?

Faisalsouz, 2023-04-11:
Does someone find any way to implement obstacle avoidance using ROS2 foxy and XRCE-DDs method?

Jaeyoung-Lim, 2023-04-11:
@Faisalsouz  https://github.com/PX4/PX4-Avoidance/pull/626

Cristian-wp, 2023-05-24:
Hello @Jaeyoung-Lim , I have read from your tagged issue that the PX4-Avoidance will be not converted from ROS1 to ROS2. So I was wondering if there are other solution to perform this.
There are other packages working well for this purpose?