PX4/PX4-Avoidance #683: Obstacle Avoidance AirSim?

hwadiadinkra, 2022-09-13:
Has anyone got the local planner working for AirSim? I read a post #114 earlier that someone did, but I am having a difficult time figuring out how to incorporate the PX4 Lidar from AirSim into the FCU frame of the local planner.

This is the transform tree without any modifications:
[image]

Things I've tried:

- Configuring a static transform between world_ned of AirSim and local_origin and fcu frame in just about every configuration:
<node pkg="tf" type="static_transform_publisher" name="tf_airsim_bridge"
          args="0 0 0 0 0 0 world_ned local_origin 10"/>

-Modify px4_config.yaml file to directly point to PX4/local_odom frame of AirSim

-Using [GazeboDrone](https://microsoft.github.io/AirSim/gazebo_drone/) which worked to some extent, but still having issues sourcing the pointcloud from AirSim into the local planner.

Any help would be appreciated.

Thanmai01, 2022-11-14:
Got any Answers for this issue? I am also facing same error

aur20, 2023-01-24:
I kinda figured out what's going on there but some gotchas remain. I suppose you use [ROS Wrapper from Microsoft](https://github.com/microsoft/AirSim/blob/main/docs/airsim_ros_pkgs.md)? Here's what I got:

- You'll need to add transforms for everything, as AirSim has it's own names and `local_planner` is not flexible. See my configuration of ```static_transforms.launch``` below. I suspect 'local_origin' corresponds to 'world' and 'fcu' to 'PX4' which is my aircraft. Further, we need tf for camera i.e. try to picture pointcloud via rviz first and you'll get an error if you don't have a tf. My camera is called 'mk1'... **Is this correct? Is there a more state of the art way to add such trivial transforms in ROS?**
```
<launch>
	<node pkg="tf" type="static_transform_publisher" name="ned_to_enu_pub" args="0 0 0 1.57 0 3.14 world_ned world_enu 100"/>
	<!-- Publish tf using 100Hz, as in PX4-Avoidance/local_planner/launch/local_planner_sitl_3cam.launch-->
	<node pkg="tf" type="static_transform_publisher" name="world_alias" args="0 0 0 0 0 0 world_ned local_origin 10"/>
	<node pkg="tf" type="static_transform_publisher" name="PX4_to_fcu_pub" args="0 0 0 0 0 0 PX4 fcu 10"/>
	<node pkg="tf" type="static_transform_publisher" name="PX4_to_mk1_pub" args="0.5 0 0.1 0 0 0 PX4 mk1_optical 10"/>
</launch>
```
- I had to launch ```local_planner```, then stop, then **start again** because the first *frame_id* from pointcloud is kept as reference, which is always empty...
- I edited `depth_image_proc` (`point_cloud_xyz.cpp` and `point_cloud_xyz_radial.cpp`) [see here](https://github.com/aur20/image_pipeline) so it will output pointcloud via timer with 15Hz because I get at most 3fps from AirSim. Maybe that was too much of good will. I cannot tell, yet. Also, consider using my launch `p2c2.launch`
- I edited `airsim_ros_wrapper.cpp` [see here](https://github.com/aur20/T3000-autonomous_drone/blob/797629f25531b4da86232284770260ca16497137/AirSim/airsim_ros_wrapper.cpp) so it will output my image in a subtopic called `image`. This is because [image_transport::subscribe_Camera](https://docs.ros.org/en/noetic/api/image_transport/html/classimage__transport_1_1ImageTransport.html) expects the *image* and *camera_info* to be subtopic of the same topic and will only query one string. Otherwise there is no pointcloud. Tell me if you fixed this more easily.

I cannot confirm correct functionality. Also I am not sure if tf used by 'fcu' is expected to be  *ENU* or *NED*.