PX4/PX4-Avoidance #687: Global planner not working

Cavalletta98, 2022-09-28:
Hi, i'm using the latest version of the code, but if i run the command `roslaunch global_planner global_planner_stereo.launch` i get this warning: `[ WARN] [1664368558.565922228, 18.000000000]:  Planner abort: missing required data `.
Can someone could help me?

Luxewave-Pongpol, 2022-11-01:
I ran into a similar problem. For my case, it seemed like [the latest commit for PX4-Avoidance](https://github.com/PX4/PX4-Avoidance/commit/4b3051707c72bf611a0ad1ca1d5e6f65b782b94f) was causing the problem. After changing to the[ previous commit](https://github.com/PX4/PX4-Avoidance/commit/303fa91b68d4d443bd8193de41b5256fa3edcb92), the global planner is working fine.

Cavalletta98, 2022-11-07:
Have you tried with fw 1.13?

Luxewave-Pongpol, 2022-11-10:
> Have you tried with fw 1.13?

Yes, v1.13.0 is also fine.

Cavalletta98, 2022-11-11:
Which ros version are you using?

Luxewave-Pongpol, 2022-11-18:
I'm using ROS melodic on Ubuntu 18.04.

Osal22, 2023-04-26:
its working with but crashes frequently any helpful insight into that  please

> I ran into a similar problem. For my case, it seemed like [the latest commit for PX4-Avoidance](https://github.com/PX4/PX4-Avoidance/commit/4b3051707c72bf611a0ad1ca1d5e6f65b782b94f) was causing the problem. After changing to [the previous commit](https://github.com/PX4/PX4-Avoidance/commit/303fa91b68d4d443bd8193de41b5256fa3edcb92), the global planner is working fine with [the firmware v1.12.3](https://github.com/PX4/PX4-Autopilot/tree/v1.12.3)