PX4/PX4-Avoidance #691: Simulate multiple iris_obs_avoid vehicles

Cavalletta98, 2022-12-27:
Hi, i'm trying to simulate multiple iris_obs_avoid vehicles. I don't understand how this line of code should be written:
`<node ns="uav0" pkg="tf" type="static_transform_publisher" name="tf_depth_camera" args="0 0 0 -1.57 0 -1.57 fcu camera_link 10"/>`
Can someone could please help me?

nicolasfripp, 2023-08-24:
Hey! @Cavalletta98  Were you able to use 'avoidance' with multiple drones? If so, could you tell me what you did? Thanks!

Cavalletta98, 2023-10-31:
Hi @nicolasfripp i'm not trying to do so anymore because this repo is working with only one drone. It should be modified a lot to work with multiple drones

nicolasfripp, 2023-11-01:
Thanks for your answer! @Cavalletta98 . Did you find any other similar package that works with multiple drones?

Cavalletta98, 2023-11-01:
Not yet