PX4/PX4-Avoidance #699: Resource not found: px4

woong137, 2023-05-15:
First of all, I am not good at English, so please understand.

I follow this. [GitHub - PX4/PX4-Avoidance: PX4 avoidance ROS node for obstacle detection and avoidance.](https://github.com/PX4/PX4-Avoidance)
But there is “Resource not found: PX4”
How can I do?

[image]

On .bashrc I wrote like this. Is this the cause of the problem?

[image]

MichaelKerley, 2023-06-07:
cd into your catkin_ws

. /opt/ros/noetic/setup.bash
. ./devel/setup.bash

aravi-2001, 2023-08-25:
source bashrc onto your  terminal by doing source ~/.bashrc (dont source devel in this case)

if even that doesnt work go to devel/setup.bash and add the same lines to you added in bashrc to devel and source devel after catkin make(dont source bashrc)

cbugk, 2024-03-28:
When I followed the readme without `CTRL-C`-ing gazebo simulation, I re-run some commands in a non-linear fashion and ended up with the same error. Solution was removing any appendage to _~/.bashrc_ file and then pasting following 5-liner present within readme:
[image]

There was not a need to add these into _devel/setup.bash_, yet @aravi-2001's comment helped me discover unintelligible lines I ended up creating. So, thx mate. @woong137, I believe you would not posses the problematic workspace at this point in time. Maybe it's best to close this issue.