PX4/PX4-Avoidance #707: Avoidance system not ready

Vu1tUrR, 2023-09-08:
Whenever I run **roslaunch local_planner local_planner_depth-camera.launch** all I get is 

WARN  [health_and_arming_checks] Preflight Fail: Avoidance system not ready

and

[ WARN] [1694162985.730929001, 45.760000000]:  Planner abort: missing required data

kerct, 2023-09-19:
Same here, I'm able to run the example with local planner stereo but not with depth camera

ChanJoon, 2023-09-21:
@Vu1tUrR @kerct 
Hi. I have experience using PX4-Avoidance recently. 
First, you need to check the topic "/camera/depth/points" in the launch file. You can find it on [line 5](https://github.com/PX4/PX4-Avoidance/blob/master/local_planner/launch/local_planner_depth-camera.launch#L5). 
If there are no `pointcloud_topics` in the planner node, those messages will be published. 

From what I remember, this problem occurred with the Gazebo model. The "iris_depth_camera" model is currently not publishing depth topics. (This might depend on your ROS version.) 
You should check which plugin is used in the model and whether it's available for your ROS version. Alternatively, you can replace the model with your custom model that includes a depth camera with iris. (If you want to use a custom model with PX4-Autopilot, you'll need to make some modifications, but that's beyond the scope of this discussion. There are many instructions available on the PX4 forum.)

kerct, 2023-09-21:
Thanks @ChanJoon!

@Vu1tUrR I solved it by changing the topic in the launch file (on line 5) to /iris_obs_avoid/camera/depth/points because I saw that the messages were being published there instead of to /camera/depth/points. I'm using Gazebo 11 on Ubuntu 20.04.

ChanJoon, 2023-09-21:
> Thanks @ChanJoon!
> 
> @Vu1tUrR I solved it by changing the topic in the launch file (on line 5) to /iris_obs_avoid/camera/depth/points because I saw that the messages were being published there instead of to /camera/depth/points. I'm using Gazebo 11 on Ubuntu 20.04.

Yes You are correct. I forgot what I did before :(
If you want to know more detailed discussions. Check out this PR https://github.com/PX4/PX4-SITL_gazebo-classic/pull/983

Mostafa12d, 2024-05-02:
Thank you very much!